xsens/xsens_dot_server #1: Node modules not found during `node-gyp rebuild`

aero-man, 2020-06-25:
I'm trying to install the Xsens DOT server on a Mac running Catalina (10.15.5). Looks like some node modules are not being found, causing the install to fail. Where can I find the `node-addon-api` and `napi-thread-safe-callback` modules?
 
Output from `npm install`:
```
$ npm install

> @abandonware/noble@1.9.2-9 install /Users/user123/Documents/ANS Lab/xsens_dot_server/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
clang: error: no such file or directory: 'Lab/xsens_dot_server/node_modules/node-addon-api'
clang: error: no such file or directory: 'Lab/xsens_dot_server/node_modules/napi-thread-safe-callback'
```

After that, there is more error output where `make` and `node-gyp` fail. See below for full output.

Here is the version info for node:
```
$ node -v
v8.9.4
$ node-gyp -v
v7.0.0
```

Full output for error:
```
$ npm install

> @abandonware/noble@1.9.2-9 install /Users/forte/Documents/ANS Lab/xsens_dot_server/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
clang: error: no such file or directory: 'Lab/xsens_dot_server/node_modules/node-addon-api'
clang: error: no such file or directory: 'Lab/xsens_dot_server/node_modules/napi-thread-safe-callback'
make: *** [Release/obj.target/binding/lib/mac/src/noble_mac.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Darwin 19.5.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/forte/Documents/ANS Lab/xsens_dot_server/node_modules/@abandonware/noble
gyp ERR! node -v v8.9.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@^0.5.3-5 (node_modules/@abandonware/noble/node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-5: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN ws@7.3.0 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@7.3.0 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself.
npm WARN xsens_dot_server@1.0.0 No repository field.
npm WARN xsens_dot_server@1.0.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/noble/node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @abandonware/noble@1.9.2-9 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @abandonware/noble@1.9.2-9 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/forte/.npm/_logs/2020-06-25T22_24_17_104Z-debug.log
```

freddijkstra, 2020-06-26:
This is very simple to solve: don't install the project in a path with spaces (your directory "ANS Lab"). That's always good advice by the way ;-)

aero-man, 2020-06-26:
Thanks. You're right. That is a simple fix. `npm install` ran without errors. `node xsensDotServer` now throws an error though. Will open a new issue.

closed 2020-06-26 by aero-man