xsens/xsens_dot_server #2: Cannot find `noble_mac` module on server startup

aero-man, 2020-06-26:
When I run `node xsensDotServer`, node throws an error that it can't find the `noble_mac` module. I am using macOS Catalina (10.15.5).

```
$ node xsensDotServer
module.js:540
    throw err;
    ^

Error: Cannot find module '../native/noble_mac'
    at Function.Module._resolveFilename (module.js:538:15)
    at Function.Module._load (module.js:468:25)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Users/user123/Documents/ANS-Lab/xsens_dot_server/node_modules/noble-mac/lib/binding.js:4:18)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
``` 

Running `npm install` right before that produces no errors, but some warnings. Here is the output of that.
```
$ npm install

> @abandonware/noble@1.9.2-9 install /Users/user123/Documents/ANS-Lab/xsens_dot_server/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
  CXX(target) Release/obj.target/binding/lib/mac/src/napi_objc.o
  CXX(target) Release/obj.target/binding/lib/mac/src/ble_manager.o
  CXX(target) Release/obj.target/binding/lib/mac/src/objc_cpp.o
  CXX(target) Release/obj.target/binding/lib/mac/src/callbacks.o
  SOLINK_MODULE(target) ../lib/mac/native/binding.node
  COPY Release/binding.node
  SOLINK_MODULE(target) Release/noble.node
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@^0.5.3-5 (node_modules/@abandonware/noble/node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-5: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN ws@7.3.0 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@7.3.0 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself.
npm WARN xsens_dot_server@1.0.0 No repository field.
npm WARN xsens_dot_server@1.0.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/noble/node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

added 2 packages in 20.497s
```

Should I install the optional dependencies? Thanks for your help.

freddijkstra, 2020-06-27:
Yes, simply renaming your directory will not do...you must remove package-lock.js and the node_modules directory. Then type `npm install` again and everything will work fine. It has been tested on MacOS Catalina without problems.

aero-man, 2020-06-27:
Thanks, Fred. Install worked.

closed 2020-06-27 by aero-man

klimstra, 2022-01-07:
Hi I am having a similar problem using Monterey on Mac OS but the above solution doesn't work

Here is the output of npm install 

$ npm install

> xpc-connection@0.1.4 install /Users/marcklimstra/xsens_dot_server/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:143:44: warning: 'GetRealNamedProperty' is deprecated [-Wdeprecated-declarations]
      Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
                                           ^
/Users/marcklimstra/.node-gyp/8.9.4/include/node/v8.h:3388:3: note: 'GetRealNamedProperty' has been explicitly marked deprecated here
  V8_DEPRECATED("Use maybe version",
  ^
/Users/marcklimstra/.node-gyp/8.9.4/include/node/v8config.h:321:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/XpcConnection.cpp:254:12: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
      Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
           ^
../../nan/nan.h:1047:3: note: 'MakeCallback' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/XpcConnection.cpp:263:12: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
      Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
           ^
../../nan/nan.h:1047:3: note: 'MakeCallback' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
  ^
../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
3 warnings generated.
  SOLINK_MODULE(target) Release/binding.node

> @abandonware/noble@1.9.2-8 install /Users/marcklimstra/xsens_dot_server/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
Traceback (most recent call last):
  File "./gyp-mac-tool", line 611, in <module>
    sys.exit(main(sys.argv[1:]))
  File "./gyp-mac-tool", line 28, in main
    exit_code = executor.Dispatch(args)
  File "./gyp-mac-tool", line 43, in Dispatch
    return getattr(self, method)(*args[1:])
  File "./gyp-mac-tool", line 246, in ExecFilterLibtool
    if not libtool_re.match(line) and not libtool_re5.match(line):
TypeError: cannot use a string pattern on a bytes-like object
make: *** [Release/nothing.a] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Darwin 21.1.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/marcklimstra/xsens_dot_server/node_modules/@abandonware/noble
gyp ERR! node -v v8.9.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN xsens_dot_server@1.0.0 No repository field.
npm WARN xsens_dot_server@1.0.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@0.5.3-6 (node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-6: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @abandonware/noble@1.9.2-8 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @abandonware/noble@1.9.2-8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/marcklimstra/.npm/_logs/2022-01-07T04_06_36_201Z-debug.log
(base) Marcs-MacBook-Pro:xsens_dot_server marcklimstra$