xsens/xsens_dot_server #4: Empty log files raspberry pi

GuidoBruinsma, 2020-07-07:
I followed the steps to setup the DOT sensors for the raspberry pi 4. Setup went without errors; DOT server is running; I am able to scan and sellect the DOTs on the localhost page. When I sellect a dot and want to start logging (and press "start logging") all seems to be working fine. A log file is generated... but, besides the headers the log files do not contain data from the DOTs. Is this a known issue?

freddijkstra, 2020-07-07:
That would only happen when the measurements were not started.
Could you check the version of the firmware? Being close by, the UT probably got one of the first sets ;-)
The sensors need to be on 1v3.

freddijkstra, 2020-07-08:
Has this issue been solved? Please close if so. Thx.

closed 2020-07-14 by freddijkstra