xsens/xsens_dot_server #5: No data generated

megautier, 2020-07-14:
Hello,

I bought a set of xsens dot sensors previous week and I'm trying to use your application to validate the data.
I tried first on ubuntu then on windows but I'm getting the same issues on both :
- Connection ok
- Start recording .... a file is generated but without data (I only have the header.

I used 2 bluetooth test : default one from my computer inter ble 4.1 and I also bought a asus-bt500 dongle to get ble 5.0. Same result with both

Do you have any idea what could be the cause?

Thank you for your help and work
Best regards,
Mélaine

freddijkstra, 2020-07-14:
Could you check the version of the firmware?

megautier, 2020-07-14:
Of course, how can I do that?

freddijkstra, 2020-07-14:
When you install the Xsens Android or iOS app and connect a sensor, you can select the sensor and see the version information.

megautier, 2020-07-14:
Ok, I'm going to look for someone with a newer phone than mine as its android version is only 6 and couldn't install the xsens application.
It is a pain that I cannot check this by another means. 

I'll keep you in touch (but I'm thinking it should be the 1.3 version as They arrived 1 week ago)

megautier, 2020-07-14:
It is 1.0 firmware.
I'm going to try to upgrade it to 1.3.0

megautier, 2020-07-14:
After the upgrade, I have the following error :
 xsensDotServer
BLE Handler started.
Web GUI Handler instantiated
Webserver listening on port 8080, IP address: 192.168.1.31

C:\Users\Jérémy\Documents\pro\micro-entreprise\ynnov\xsens_dot_server>node xsensDotServer
BLE Handler started.
Web GUI Handler instantiated
Webserver listening on port 8080, IP address: 192.168.1.31
C:\Users\Jérémy\Documents\pro\micro-entreprise\ynnov\xsens_dot_server\node_modules\noble\lib\hci-socket\hci.js:582
    this.address = result.toString('hex').match(/.{1,2}/g).reverse().join(':');
                                                          ^

TypeError: Cannot read property 'reverse' of null
    at Hci.processCmdCompleteEvent (C:\Users\Jérémy\Documents\pro\micro-entreprise\ynnov\xsens_dot_server\node_modules\noble\lib\hci-socket\hci.js:582:59)
    at Hci.onSocketData (C:\Users\Jérémy\Documents\pro\micro-entreprise\ynnov\xsens_dot_server\node_modules\noble\lib\hci-socket\hci.js:461:12)
    at BluetoothHciSocket.emit (events.js:310:20)
    at BluetoothHciSocket.onHciEventEndpointData (C:\Users\Jérémy\Documents\pro\micro-entreprise\ynnov\xsens_dot_server\node_modules\bluetooth-hci-socket\lib\usb.js:156:10)
    at InEndpoint.emit (events.js:310:20)
    at Transfer.transferDone (C:\Users\Jérémy\Documents\pro\micro-entreprise\ynnov\xsens_dot_server\node_modules\usb\usb.js:441:9)

freddijkstra, 2020-07-14:
Closed because of wrong firmware version.

closed 2020-07-14 by freddijkstra