xsens/xsens_dot_server #7: Connection to Xsens DOTs from other languages

aero-man, 2020-09-23:
Hi, I am trying to connect to the Xsens DOTs using Python and Bluepy. I have a general question about enabling and receiving notifications from the Xsens DOTs. Am I correct that to get accel/gyro data out of the sensor, I have to write an enable message to the Control Characteristic (`b\x01\x01\02` for example), then one of the measurement characteristics (short/medium payload, etc.) will send out notification packets with sensor data?

I am able to connect to and write to the control characteristic of an Xsens DOT sensor from my Python script but am having trouble getting notifications. Do I have the general idea of getting sensor data correct?

Thanks.

aero-man, 2020-09-30:
I was told by Xsens Support over email that prior to writing to the control characteristic, I have to "write something like 0x2003 and 0x2004 characteristics to enable them specifically. This was from Alessandro. Do you know what he means by this? Are there specific lines in the Node server that accomplish this? Thanks.

aero-man, 2020-10-01:
I fixed the issue. Writing it here in case anybody else wants to connect to the sensors via Python and [Bluepy](https://github.com/IanHarvey/bluepy/tree/master/bluepy) on Linux.

The steps for getting data from the sensors are:

1. Scan for the sensor (can skip this if you already know the Bluetooth address)
2. Connect to the sensor using the Bluetooth address found via scanning
3. Subscribe to the Measurement Characteristic by writing an enable message to that characteristic's CCCD, which is at integer handle + 1 (e.g. if the characteristic's handle is 39, write to it's CCCD at handle 40) (There are two measurement characteristics to choose from for Xsens DOTs, depending on whether your measurement type is a short or medium payload)
4. Write an enable message to the Control Characteristic
5. Add a delegate to your sensor to start listening for notifications from the sensor
6. Run a loop to listen for notifications and do something with the data received. (The example below just prints it.)

Example code (skips the first sensor scanning step):
```
'''
Example code for getting data from a single Xsens DOT sensor in Python (with Bluepy and Raspbian OS)

Python 3.7.3
bluepy 1.3.0
Raspbian GNU/Linux 10
'''

from bluepy import btle

'''
2. Connect to the sensor using the Bluetooth address
'''
bluetooth_address = "d4:ca:6e:12:34:56" # YOUR XSENS DOT BLUETOOTH ADDRESS HERE
p = btle.Peripheral(bluetooth_address)

'''
3. Subscribe to the appropriate characteristic for the payload we want by enabling its CCCD. 
The CCCD's handle is the characteristic handle + 1.
In this case, we want to measure in "Extended (Quaternion)" mode. According to Xsens's docs, 
this uses the Medium Payload Length Characteristic. Bluepy's valHandle() says this characteristic's 
integer handle is 39. Bluepy does not always recognize `0x2000` style handles.
'''
medium_payload_length_characteristic_handle = 39
cccd_handle = medium_payload_length_characteristic_handle + 1
CCCD_ENABLE_MESSAGE = b"\x01\x00"
p.writeCharacteristic(cccd_handle, CCCD_ENABLE_MESSAGE)

'''
4. Enable the Xsens DOT sensor by writing the enable message to the Control characteristic, 
whose integer handle is `32`. The last `\x02` in `ENABLE_MESSAGE` specifies that we want
"Extended (Quaternion)" mode.
'''
ENABLE_MESSAGE = b"\x01\x01\x02"
p.writeCharacteristic(32, ENABLE_MESSAGE)

'''
5. Add a delegate to your sensor object (`p`) to allow notifications from the sensor.
The custom delegate below inherits bluepy's DefaultDelegate so we can override the handleNotification()
callback function and print the data we receive.
'''
class XsensNotificationDelegate(btle.DefaultDelegate):
    def __init__(self):
        btle.DefaultDelegate.__init__(self)

    # Do something when sensor data is received. In this case, print it out.
    def handleNotification(self, cHandle, data):
        print("Received notification from cHandle: {0} \n Data received: {1}".format(cHandle, data))

p.setDelegate(XsensNotificationDelegate())

'''
6. Start listening for sensor data
The callback function handleNotification() written above will be called once a notification is received.
'''
while True:
    if p.waitForNotifications(1.0): # Listen for 1.0 second for a notification before giving up
        print("handleNotifications was called")
    else:
        print("No notification from the Xsens DOT sensor was received")
```

closed 2020-10-01 by aero-man

FrederikPetri, 2020-10-22:
@aero-man
Hi,
Very useful python script, thanks for sharing! Can you explain how you convert the bytes to sensor data (timestamp and quaternions)?