xsens/xsens_dot_server #8: Cannot find module 'binding.node' on Raspberry Pi

aero-man, 2020-09-29:
I am getting the following error on Raspberry Pi while trying to start the latest version of the server. Previous versions of the server worked fine on this Pi running the same version of Raspberry Pi OS. I ran `npm install` and `sudo npm install` again to troubleshoot, but they didn't fix the issue.

```
$ sudo node xsensDotServer
module.js:549
    throw err;
    ^

Error: Cannot find module '../build/Release/binding.node'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Documents/ANS-Lab/xsens_dot_server/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
```

Specs:
node -v = v8.11.1
Raspberry Pi 3 Model B+
1 GB RAM
Raspberry Pi OS (Raspbian), Version: May 2020, Kernel: 4.19

gnixob, 2020-09-30:
Hi, 

Which firmware version are you using? The latest server only support firmware 1.4.0 since the synchronization function has been added.

fancycww, 2020-10-13:
Hi,
It might be the environment issue.

Please see the post, it could help.
https://base.xsens.com/hc/en-us/community/posts/360009645740-Can-t-start-measurement-with-Noble?page=1#community_comment_360002190000