xsens/xsens_dot_server #9: Error when starting scanning

FrederikPetri, 2020-10-01:
I get the error: "ERROR: component 'SensorServer' received unexpected event 'startScanning' in state 'Powering-on'" when starting scanning for the sensors. Can you help me solve this?

freddijkstra, 2020-10-01:
Enable your Bluetooth?

FrederikPetri, 2020-10-01:
Hi Fred,

I followed the guide and changed my Bluetooth driver to WinUSB. It should be enabled when I look it up in Device Manager?
[image]

freddijkstra, 2020-10-01:
As I'm not familiar with Windows (I use Mac), I unfortunately cannot give you advise on that area. However, the error occurs when the Bluetooth adapter is not enabled (I agree that the error is not descriptive :-). So either the installation failed, the wrong adapter is used or some other Windows related problem persists.

closed 2020-10-01 by FrederikPetri