xsens/xsens_dot_server #13: Nuber of frames from each sensor does not match

Motioncatch, 2020-10-15:
Hi,  
I am using the Xsens dot server to record and log quaternions from the Xsens dots. I have connected 5 sensors, they are being synced and start logging. When I review the .csv log file afterward the number of frames from each sensor does not match:
Sensor 1: 1308 frames
Sensor 2: 639 frames
Sensor 3: 645 frames
Sensor 4: 626 frames
Sensor 5: 1319 frames
 I have tried multiple times, and there is always 1 or 2 sensors having more than double the number of samples than the others. 
 I running the server on a raspberry py 4 as suggested. 
 Best regards Mathias

gnixob, 2020-10-16:
Hi Motioncatch,

It should be the data loss if some sensor has less frames. We have received feedback that synchronization will increase the data loss when using Xsens DOT server. You can try without the synchronization and see how it goes.

Motioncatch, 2020-10-16:
Hi gmixob 

Thanks for the rapid reply. I have now tried multiple times without synchronization.

Motioncatch, 2020-10-16:
(I closed the issue by mistake, still very open)
Hi gmixob

Thanks for the reply. I have now tried multiple times without synchronization, and the result is the same. Sometimes 3 or 4 sensors get the same amount of samples, but there is always one sensor with a data loss.  
I have also tried with the official Xsens dot app without synchronization and the result is also the same, some sensors have a data loss. 
Any way this can be improved?
I think this is a major issue there should be improved.

freddijkstra, 2020-10-16:
Hi Motioncatch,

We (that is [ORYX Movement Solutions)](https://www.oryxmovementsolutions.com/) have a solution built on a Raspberry Pi 3 as well using 8 sensors. Our application does not experience any data-loss - that is when the sensors are close enough of course. So, it is possible and I'm wondering what is different in your situation. I also wonder why you see data-loss in the Xsens DOT app. Do you still have the sensors placed in the charger?

Motioncatch, 2020-10-19:
Hi Fred, 
Thanks for your comment. 
I had the sensors placed in the charger during the testing. I have now tried to take them out of the charger and it fixed the issue with the Xsens DOT app. However, I keep getting the data-loss using the Xsens dot server on raspberry pi. 
The only difference I can see is I'm using the Raspberry Pi 4 Model B 4GB RAM compared to the Raspberry pi 3.

freddijkstra, 2020-10-19:
Ok, so we're halfway there ;-)
Just to check to be sure: do you have other Bluetooth devices connected to the Raspberry? Like a wireless mouse or keyboard?

Motioncatch, 2020-10-20:
Hi Fred, thanks for your help.  
I have been using a Bluetooth mouse with its own dongle. I have tried with a wired mouse instead now. The measurements get more stable. However, 1 sensor still having a data loss of approximately 50% compared to the rest. The result is the same with or without sync. The sensor with the data loss is shifting from sensor to sensor at each trial.

Motioncatch, 2020-10-20:
When I'm using 8 sensors as you mentioned I have 2-4 sensors with data-loss.

Motioncatch, 2020-10-23:
Hi @freddijkstra. 
I still have the data-loss issue and it is frustrating me. You previously said you didn't have any problems with using the Raspberry pi and the Xsens dot server, so I wonder the difference in our setups. 
My setup: 
Raspberry pi 4 with raspbian 
Running the original sensor dot server
Using the internal Bluetooth adapter on the raspberry (Bluetooth version 5.0)

I will be very grateful if you would share how you have set up your system?