xsens/xsens_dot_server #14: Unable to connect multiple dots in Raspberry pi 4

naveenkumar2095, 2020-10-21:
Hello,

I am trying to connect multiple dots at same time using raspberry pi 4 model but unable to achieve this task. due to some issue, one sensor connect at a time and rest of the sensors  continuously  showing  "stop connecting " in the GUI. let me how to rectify this problem.

Best Regards
R Naveen kumar 
NXtQ Pvt Ltd
[image]
[image]

fancycww, 2020-10-21:
Hi,
It seems that you used the old firmware 1.0.0.
Could you please check the version of the firmware?
Please use the latest version 1.4.0.
Thanks.

naveenkumar2095, 2020-10-21:
Hello

Thank you for your response. Ya checked in the readme file it's mentioned
version 1.4.0

Regards
Naveen Kumar
NxtQ Pvt Ltd

naveenkumar2095, 2020-10-21:
Please find the screenshots attached in the github issues it is
clearly showing code issue in blehandler.js
Best Regards
R Naveen Kumar

naveenkumar2095, 2020-10-21:
If it is failure because of firmware version, Can you please send me the procedure to upgrade the version to 1.4.0?

gnixob, 2020-10-22:
> If it is failure because of firmware version, Can you please send me the procedure to upgrade the version to 1.4.0?

Hi, please follow these steps:
1. Download "Xsens DOT" app from Google play store or App store
2. Connect the senors
3. Go to "Advance - Device Firmware Upgrade"
4. Follow the instructions in the app to do the upgrade.

Thanks!

naveenkumar2095, 2020-10-22:
Thank you for your response 

Best regards 
Naveen kumar
NxtQ Pvt Ltd

closed 2020-10-22 by gnixob