xsens/xsens_dot_server #16: Unable to read service characteristics on Mac

njanssen, 2020-10-24:
I'm running into the known issue where it's not possible to connect to a DOT using Mac

njanssen, 2020-10-24:
I did some additional testing to make sure the Node.js and BLE setup on my MacBook is correct. I can successfully discover services and characteristics on my Arduino Nano 33 BLE that runs the standard [BatteryMonitor example in the ArduinoBLE library](https://github.com/arduino-libraries/ArduinoBLE/tree/master/examples/Peripheral/BatteryMonitor). With this device, discovery also works with the Light Blue app.

Strange that this works perfectly fine, but I can't discover characteristics of my Xsens DOT sensors on macOS .. Thoughts?

freddijkstra, 2020-10-24:
It seems that newer MACs with Bluetooth 5 could be the problem. I have one Mac with Bluetooth 4.2 with no problem. However on my new macbook pro with Bluetooth 5 I have the same problem you have.

njanssen, 2020-10-25:
Hi @freddijkstra, I experience these connection issues on a MacBook Pro (2017, 15") with a built-in Broadcom Bluetooth 4.2 adapter. So, I don't think there's a direct relation between these issues and built-in adapters that support Bluetooth 5. I'm running macOS Catalina (10.15.17) and my DOTs are using the latest firmware (1.4.0). Is your older Mac perhaps using an older macOS version or different Bluetooth radio chip? Might be worth comparing. 

Since the problem also occurs outside Node.js (e.g. when using the Lightblue app), it seems to me that macOS Catalina's CoreBluetooth framework and the DOT firmware have compatibility issues. Please note that my MacBook can connect to other Bluetooth devices without issues. 

Is this incompatibility problem already being addressed within Xsens?

freddijkstra, 2020-10-25:
Hi Niels, thanks for looking into this. You're correct. I tested it with the LightBlue app as well and see that the Xsens DOT will connect using my old MAC (Broadcom, chipset 20702B0 Bluetooth Core Spec 4.0) and not with my new MAC (Broadcom, chipset 4364B3 Bluetooth Core Spec 5.0). I will pass this on to the Xsens team so they can look into this.

njanssen, 2020-10-25:
> I will pass this on to the Xsens team so they can look into this.

Thanks Fred!

riasatislam, 2020-11-06:
@njanssen I am having similar problems as you. Seems like same device spec as well, I have a MacBook Pro (2017, 15") with a built-in Bluetooth 4.2.

njanssen, 2020-11-09:
> @njanssen I am having similar problems as you. Seems like same device spec as well, I have a MacBook Pro (2017, 15") with a built-in Bluetooth 4.2.

@riasatislam Thanks for sharing! Wondering .. which macOS version are you using? I'm running the latest version of Catalina.

@freddijkstra Which versions are on your older Mac (with working connectivity) and on your newer Mac (that has the same issue)? Maybe that gives us a lead that could (hopefully) help the Xsens DOT team find the root cause of our issue.

riasatislam, 2020-11-10:
I am on the lastest Catalina too, 10.15.7

yusheng-jia, 2020-11-11:
@njanssen Thanks for sharing the information about this. You know that we are use the noble to manage the ble services and characteristics. We have the same issue with Mac 2019 (10.15.7), we think that there is something wrong on Bluetooth 5.0, and the noble does not adapt to the latest version of Mac. We put up a issue to noble and hope they give some new update.
But anyway if want to solve this issue , we must familiar with the native  implementation like out Xsens DOT on mobile phone. We are looking into this and it will take some time.

freddijkstra, 2020-11-11:
> @njanssen Thanks for sharing the information about this. You know that we are use the noble to manage the ble services and characteristics. We have the same issue with MacBook Pro 2019 (10.15.7), we think that there is something wrong on Bluetooth 5.0, and the noble does not adapt to the latest version of Mac. We put up a issue to noble and hope they give some new update.
> But anyway if want to solve this issue , we must familiar with the native implementation like our Xsens DOT on mobile phone. We are looking into this and it will take some time.

I don't necessarily agree with you conclusion. Just using the LightBlue app also shows that the Xsens DOT disconnects while being interrogated (see attached screenshot). As such I don't think that this necessarily has something to do with Noble.
[image]

yusheng-jia, 2020-11-11:
@freddijkstra Thanks , My still puzzled is why it works on Bluetooth 4.2 and didn't work on Bluetooth 5.0. If no modification on the noble library or Light blue app, We need to do research on Xsens DOT's firmware or something else?

njanssen, 2020-11-11:
@yusheng-jia Please note that the Bluetooth adapter is my MacBook (which doesn't connect to the DOT) has Bluetooth 4.2. Same goes for @riasatislam. So the problem also doesn't seem to be bound to Bluetooth 5.0. And @freddijkstra is also right about the problem being independent of Noble, I experience the same results with a native app such as LightBlue,

njanssen, 2020-12-07:
Hi @yusheng-jia, @freddijkstra, it's been a while since the last update. Has the Xsens DOT team been able to make some progress on this macOS incompatibility issue?

fancycww, 2020-12-08:
Hi @njanssen,
We have solved this with the new FW v1.6 which will be released on Dec 15, 2020.

It's an issue within OSx.
MacBook does not accept `LL_PHY_REQ` message. Upon reception it appears that the MacBook stops communicating, making the connection time out on the DOT side.

Since Apple may not fix this in short time, so we made a change in the DOT firmware:
`LL_PHY_REQ` is removed from the firmware.

We test it on MacBook, it works. Please try out when the new firmware is available.

Thanks.

njanssen, 2020-12-08:
That's great news! Thanks for the update @fancycww, I'll check out the new firmware as soon as it's released 👍

njanssen, 2020-12-17:
Perfect, I can confirm that the new firmware version (1.6.0) resolves this issue. I was able to connect and use the DOT sensors on my Mac using my Node.js library without further changes: https://www.npmjs.com/package/@vliegwerk/xsens-dot

Thanks everyone for the support!

closed 2020-12-17 by njanssen