xsens/xsens_dot_server #19: unable to get log file from 5 sensors at a time using R Pi 4 model

naveenkumar2095, 2020-11-06:
Hi

I connect all 5 sensors using raspberry pi 4 model , but unable to get the log files from all dots at same time. some time this issue is coming for me . not all the time. once i restart the pi then it will work properly , after 5 to 6 min it will go to same state , automatically out of 5 one dot is disconnecting . please let me know how to solve this problem.

Best regards 
Naveen kumar 
NxtQ Pvt Ltd

naveenkumar2095, 2020-11-06:
[image]
[image]

naveenkumar2095, 2020-11-06:
[image]

naveenkumar2095, 2020-11-06:
Note: All dots are powered up using case

freddijkstra, 2020-11-06:
> Note: All dots are powered up using case

You probably want to test this also when the sensors are not powered. There is a difference in behaviour by our experience.

freddijkstra, 2020-11-06:
The EADDRINUSE error occurs when the previous script did not properly finish. The port is not released by the OS. Restart the Pi is the easiest option.