xsens/xsens_dot_server #22: Xsens Dot not connecting to the Server WebApp

dt1111, 2020-12-14:
Hello,

I've been trying to connect the xsens dot to the webapp server unsuccessfully. I'm able to scan the sensors successfully. But I haven't managed to connect them all. Only once I was able to connect 1 sensor but then when I clicked in the button to gather data, the server app didn't react. 

I'm running this in a windows 10 computer

Have you encountered this problem? If yes, how did you solve it?

Thank you

naveenkumar2095, 2020-12-21:
upgrade dots using xsens app

dt1111, 2020-12-21:
Thank you, I already did it and sent an email to support showing in a video how the problem still persists. Any additional ideas?

gnixob, 2020-12-24:
Hi, our support team have not received your video, could you send again or maybe just add some screenshots here. Thanks!

dt1111, 2020-12-24:
I sent it to Alessandro, who saw it (you might want to discuss it with him). Otherwise, could you provide me with an email so that I can send it to you again? Thanks!

gnixob, 2020-12-25:
Sorry but Alessandro is on holidays. With the new function in GitHub, you can directly drag and drop the video here.