xsens/xsens_dot_server #23: Issues when connecting Xsens DOT

LuzyCat, 2021-01-05:
Hello,

I have used xsens_dot_server for Xsens DOT in windows 10.
There is a problem that cannot be connected as in the previous issue. And I got a message that
`ERROR: component 'SensorServer' received unexpected event 'bleSensorData' in state 'Idle' `
To solve this, I checked the sensorServer.js code and created a new state diagram(cause it's completely different from the document).
[image]

1. Considering this, the process of making the sensor idle state after checking whether the sensor is disconnected is repeated. In this case, why is the event 'bleSensorData' occurring?
2. And I think when the connection of the sensor is temporarily disconnected, it would be better to try to reconnect it. Now, even though it returns to the idle state, the web GUI does not change to the disconnected button. This is very confusing.
3. Besides, I have to connect at least 6 sensors simultaneously, but it is very difficult to connect more than 3 sensors... I wonder if it worked well when you experimented with multiple connections.

fancycww, 2021-01-05:
Hi,

This code base is going to be soon obsolete, we will provide a new QT project instead of this, we will update soon.

In this project:
We changed the design, but we didn't update the document, sorry about this and thank you for the new diagram.

In previous version, after selecting the sensors, you can connect them all at once and then disconnect the sensors selected before.
And current version, each sensor can be connected / disconnected separately, so there is no global connection / disconnection state.
 
So the state machine is more complex than before, and the code is not very elegant.
 
About the questions:
1.	Considering this, the process of making the sensor idle state after checking whether the sensor is disconnected is repeated. In this case, why is the event 'bleSensorData' occurring?
 
I'm not sure about your operation process, but this kind of error is possible.
For example, a sensor is reporting data all the time. At this time, stopping logging returns to idle state, but there is still data in BLE cache. However, in idle state, it no longer accepts data and writes it to the CSV file (the CSV file has been closed). This error message will appear.
 
2.	And I think when the connection of the sensor is temporarily disconnected, it would be better to try to reconnect it. Now, even though it returns to the idle state, the web GUI does not change to the disconnected button. This is very confusing.
 
At present, we don't have a reconnection mechanism in this project. In the previous version, I managed the connection state uniformly, but in the later version, we designed that each sensor can be connected / disconnected separately. Therefore, idle state is a global state, not for maintaining each sensor. When a sensor is connected, it can also be idle state, because the sensor does not perform real-time streaming and other operations.
In idle state, some sensors are connected and some are not.
 
3.	Besides, I have to connect at least 6 sensors simultaneously, but it is very difficult to connect more than 3 sensors... I wonder if it worked well when you experimented with multiple connections.
 
We can connect about 5 sensors on several computers.
But some computers can't connect to any of them, which depends on the Noble Library and BLE hardware adapters.
 
Here are our test results:
[image]

[image]

Thanks.

LuzyCat, 2021-01-06:
Thank you for your comment.

In my case, I could connect 6 sensors first, but most of the connections were lost when I started logging, whether synchronization was turned on or off.

Here is my result:
[image]

In the table you sent, does the meaning of 'stable' means that the csv file with all sensor data recorded is stably generated?

fancycww, 2021-01-07:
Hi Meejin,
'Stable' means that the connection status is stable.

LuzyCat, 2021-01-07:
Hi Weiwei,

Ok.
If this issue is caused by the limitation on the number of simultaneous connections supported by the Noble Library, there is no way to solve it so far. Thanks for your help.

closed 2023-05-31 by LuzyCat