xsens/xsens_dot_server #31: TypeError: Cannot read property 'reverse' of null

FakeItUntilYaMakeIt, 2021-06-10:
Hi there,
I'm new to development so please forgive me if this is a dumb question - I have followed the instructions to install and run the xsens dot server and I am getting the error below (all output is included). Please let me know how to troubleshoot. Thank you!

C:\Users\mkamachi\xsens_dot_server>node xsensDotServer
Web GUI Handler instantiated
BLE Handler started.
Webserver listening on port 8080, IP address: 192.168.1.80
C:\Users\mkamachi\xsens_dot_server\node_modules\noble\lib\hci-socket\hci.js:582
    this.address = result.toString('hex').match(/.{1,2}/g).reverse().join(':');
                                                          ^

TypeError: Cannot read property 'reverse' of null
    at Hci.processCmdCompleteEvent (C:\Users\mkamachi\xsens_dot_server\node_modules\noble\lib\hci-socket\hci.js:582:59)
    at Hci.onSocketData (C:\Users\mkamachi\xsens_dot_server\node_modules\noble\lib\hci-socket\hci.js:461:12)
    at BluetoothHciSocket.emit (events.js:310:20)
    at BluetoothHciSocket.onHciEventEndpointData (C:\Users\mkamachi\xsens_dot_server\node_modules\bluetooth-hci-socket\lib\usb.js:156:10)
    at InEndpoint.emit (events.js:310:20)
    at Transfer.transferDone (C:\Users\mkamachi\xsens_dot_server\node_modules\usb\usb.js:441:9)

FakeItUntilYaMakeIt, 2021-06-18:
More details:

C:\Users\mkamachi\xsens_dot_server>node xsensDotServer
Web GUI Handler instantiated
BLE Handler started.
Webserver listening on port 8080, IP address: 192.168.1.82
FATAL ERROR:  Attempt to access memory outside buffer bounds
 1: 00007FF77E96CF2F napi_wrap+112799
 2: 00007FF77E90CF26 v8::base::CPU::has_sse+55702
 3: 00007FF77E90DDB3 v8::base::CPU::has_sse+59427
 4: 00007FF77E90D509 v8::base::CPU::has_sse+57209
 5: 00007FF77E932410 napi_fatal_error+160
 6: 00007FFD910C00C7
 7: 00007FFD91175A36
 8: 00007FFD9110A740
 9: 00007FFD910FDB49
10: 00007FFD9AF31506 RtlCaptureContext2+1190
11: 00007FFD910D715E
12: 00007FFD910BFA58
13: 00007FF77E930902 node_module_register+3042
14: 00007FF77E9AFBF8 uv_idle_init+120
15: 00007FF77E9B983E uv_run+254
16: 00007FF77E8C9681 v8::internal::interpreter::BytecodeArrayWriter::source_position_table_builder+31713
17: 00007FF77E936223 node::Start+275
18: 00007FF77E7B6A9C RC4_options+340380
19: 00007FF77F62F3F8 v8::internal::SetupIsolateDelegate::SetupHeap+1300536
20: 00007FFD99207034 BaseThreadInitThunk+20
21: 00007FFD9AEE2651 RtlUserThreadStart+33

fancycww, 2021-06-29:
Hi,
From the log, there is an error in the dependent library HCI socket.
What I can think of is, have you add Bluetooth vid & PID in usb.js? 
Thanks.

FakeItUntilYaMakeIt, 2021-06-29:
I did. Please see below photos. Was this done correctly?

[image]
[image]

FakeItUntilYaMakeIt, 2021-06-29:
It now appears to be running. I updated the npm debug package. Now is it continues to be in the "Powering-on" state. When I click "start scanning" I receive the error "ERROR: component 'SensorServer' received unexpected event 'startScanning' in state 'Powering-on'"

FakeItUntilYaMakeIt, 2021-06-29:
The firmware of my sensors is 1.6.0, would that be causing a problem? It says it is unable to downgrade.

kap7475, 2022-08-05:
I'm having the same issues, however I still get the reverse of null error happen every now and then, and also a 'Buffer is out of bounds' error and now I too am stuck with the powering on state. Yesterday I was still able to connect to the server properly if I kept closing the cmd and trying to run it again but now I cant seem to get it to work at all. I did reinstall the node an npm packages but like I said it was still working, but now its suddenly not.