xsens/xsens_dot_server #34: Error installing xsens Dot Server

Allwillcome, 2021-09-10:
- OS name: Microsoft Windows 10 Home Chinese Version

- OS version: 10.0.19043 Not available Build 19043

- OS manufacturer: Microsoft Corporation

- Bluetooth version: LMP 8.4096 Bluetooth 4.2

[image]
## Install the following tools
```
conda create -n XsensDotServe python=3.8.3
```
- I have successfully registered npm and node-gyp
- An error occurs when I run Windows-build-tools in CMD as administrator

```
npm WARN deprecated request@2.88.2: request has been deprecated, see
https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN rollback Rolling back windows-build-tools@5.2.2 failed (this is probably
harmless): EBUSY: resource busy or locked, rmdir
'C:\Users\openmind\AppData\Roaming\npm\node_modules\windows-build-tools'
npm ERR! code EBUSY
npm ERR! syscall rename
npm ERR! path C:\Users\openmind\AppData\Roaming\npm\node_modules\windowsbuild-tools
npm ERR! dest C:\Users\openmind\AppData\Roaming\npm\node_modules\.windowsbuild-tools.DELETE
npm ERR! errno -4082
npm ERR! EBUSY: resource busy or locked, rename
'C:\Users\openmind\AppData\Roaming\npm\node_modules\windows-build-tools' ->
'C:\Users\openmind\AppData\Roaming\npm\node_modules\.windows-buildtools.DELETE'
npm ERR! A complete log of this run can be found in:
npm ERR! C:\Users\openmind\AppData\Roaming\npm-cache\_logs\2021-09-
04T03_36_56_161Z-debug.log
```
Is windows-build-Tools not working?

I successfully registered with Zadig

[image]
## Set Up on Windows
I have cloned the repository and have successfully registered NPM when I run NPM Install
```powershell
npm WARN xsens_dot_server@1.0.0 No repository field.
npm WARN xsens_dot_server@1.0.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4
(node_modules\xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpcconnection@0.1.4: wanted {"os":"darwin","arch":"any"} (current:
{"os":"win32","arch":"x64"})
audited 212 packages in 2.75s
3 packages are looking for funding
run `npm fund` for details
found 11 vulnerabilities (1 low, 10 high)
run `npm audit fix` to fix them, or `npm audit` for details
```
- Run Xsens Dot Server
```powershell
Web GUI Handler instantiated
Webserver listening on port 8080, IP address: 192.168.31.72
1
2
E:\Workspace\xsensdot\xsens_dot_server\node_modules\bluetooth-hcisocket\lib\usb.js:70
throw new Error('No compatible USB Bluetooth 4.0 device found!');
^
Error: No compatible USB Bluetooth 4.0 device found!
at BluetoothHciSocket.bindUser
(E:\Workspace\xsensdot\xsens_dot_server\node_modules\bluetooth-hcisocket\lib\usb.js:70:11)
at BluetoothHciSocket.bindRaw
(E:\Workspace\xsensdot\xsens_dot_server\node_modules\bluetooth-hcisocket\lib\usb.js:28:8)
at Hci.init (E:\Workspace\xsensdot\xsens_dot_server\node_modules\noble\lib\hcisocket\hci.js:101:35)
at NobleBindings.init
(E:\Workspace\xsensdot\xsens_dot_server\node_modules\noble\lib\hcisocket\bindings.js:82:13)
at processTicksAndRejections (internal/process/task_queues.js:79:11)
(XsensDotServe) PS E:\Workspace\xsensdot\xsens_dot_server> npm install
npm WARN xsens_dot_server@1.0.0 No repository field.
npm WARN xsens_dot_server@1.0.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4
(node_modules\xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpcconnection@0.1.4: wanted {"os":"darwin","arch":"any"} (current:
{"os":"win32","arch":"x64"})
audited 212 packages in 2.75s
3 packages are looking for funding
run `npm fund` for details
found 11 vulnerabilities (1 low, 10 high)
run `npm audit fix` to fix them, or `npm audit` for details

```

AitorIriondo, 2021-09-15:
Hello!

I got the same issue in Windows the first time. Have you tried the following troubleshooting?:

Add Bluetooth adapter
If you encounter Error: No compatible USB Bluetooth 4.0 device found! when try to run Xsens DOT Server on Windows, it means you need to add your Bluetooth adapter to the USB device list:

Open Device Manager, find the VID and PID of your Bluetooth adapter.
 
Open source code: xsens_dot_server\node_modules\bluetooth-hci-socket\lib\usb.js
Add Bluetooth VID & PID in usb.js (line 66), save and close.
 
Run Xsens DOT Server again.

Allwillcome, 2021-09-17:
I have added Bluetooth VID & PID in usb.js (line 66) save and close
[image]

```powershell
this._usbDevice = usb.findByIds(0x8087, 0x0A2A) || usb.findByIds(0x8087, 0x0A2A) || usb.findByIds(0x8087, 0x0A2A) || usb.findByIds(0x8087, 0x0A2A) || usb.findByIds(0x8087, 0x0A2A);
```

But still failed after running node xsensDotServer

```
Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (E:\Workspace\xsensdot\xsens_dot_server\node_modules\_bluetooth-hci-socket@0.5.1@bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (E:\Workspace\xsensdot\xsens_dot_server\node_modules\_bluetooth-hci-socket@0.5.1@bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (E:\Workspace\xsensdot\xsens_dot_server\node_modules\_noble@1.9.1@noble\lib\hci-socket\hci.js:101:35)
    at NobleBindings.init (E:\Workspace\xsensdot\xsens_dot_server\node_modules\_noble@1.9.1@noble\lib\hci-socket\bindings.js:82:13)
    at Noble.<anonymous> (E:\Workspace\xsensdot\xsens_dot_server\node_modules\_noble@1.9.1@noble\lib\noble.js:57:24)
    at processTicksAndRejections (internal/process/task_queues.js:79:11)
```

AitorIriondo, 2021-09-17:
Have you tried to reinstall by using Zadig? You can go to device manager, update the driver you created by using Zadig, and it should go back to normal. Then you install again Zadig and hopefully will work. I had to do this 3 times before it worked. Also, Zadig fails quite a lot, so even if it said successfully installed I had to rerun it twice more.

Allwillcome, 2021-09-18:
After several attempts, I finally managed to run Xsens Dot Server successfully. Thank you for your patient guidance.Thank you!
[image]

closed 2021-09-18 by Allwillcome